pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #1: error

platone58, 2020-05-28:
can you help me?
i have a error compiling . this 
'class SI4735' has no member named 'seekStationProgress'
but i have set libraryes and also all dependancies

pe0mgb, 2020-05-29:
Hello, I have compiled the sketch with the latest version of the library , downloaded from de si4735 github from PU2CLR Ricardo. I did not find any problem. Compilation and linking ran 100% OK. So check you library. Do have maybe the library installed twice?

 

Regards,

Gert PE0MGB

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

platone58, 2020-05-29:
I deleted and reinstalled the last library for Si4735 and the error is gone
but now I have this error:
exit status 1
Error compiling for board ESP32 Dev Module. I think my ESP32 library is not
good, which library did you use? thanks Giovanni from Italy

platone58, 2020-05-29:
thanks , i have deleted all library and reinstalled , now work whitout error

closed 2020-05-29 by platone58