pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #2: When I compile the sketch I receive error

Snowist, 2020-05-31:
I think following function needs definition before called
--------------------------------------------------------------------------------

si4735_2.8_TFT:717:16: error: expected initializer before 'RotaryEncFreq'

 void IRAM_ATTR RotaryEncFreq()

Snowist, 2020-05-31:
Solved. Wrong board selected. Currently selected ESP32 Dev board. However new compile error appeared. I have deleted other encode libraries, but this error still persist.

C:\Users\snowb\Documents\Arduino\Projeler\SI4735-Radio-ESP32-Touchscreen-Arduino-master\si4735_2.8_TFT\si4735_2.8_TFT.ino: In function 'void setup()':

si4735_2.8_TFT:779:57: error: 'RotaryEncFreq' was not declared in this scope

   attachInterrupt(digitalPinToInterrupt(ENCODER_PIN_A), RotaryEncFreq, CHANGE);

pe0mgb, 2020-05-31:
Must be wrong library. Use the supplied encoder library.

Outlook voor Android downloaden

Snowist, 2020-06-01:
There is no other Rotary library. Just  PU2CLR_SI4735 and TFT_eSPI libs. Also tried this in VS Code Platform Tools and same error happened.

Snowist, 2020-06-01:
Found missing declaration of "RotaryEncFreq" in code and addded and created pull request.

Snowist, 2020-06-01:
I tried to add following code in my in, but just realized it is already available in .ino file here. I didn't understand why my download is missing this part of code. When I add this code worked fine.

//=======================================================================================
void IRAM_ATTR RotaryEncFreq()
//=======================================================================================
{ 
  // rotary encoder events
  if (!writingEeprom){
    encoderStatus = encoder.process();
    
    if (encoderStatus)
    {
      if (encoderStatus == DIR_CW)// Direction clockwise
      {
        encoderCount = 1;
      }
      else
      {
        encoderCount = -1;
      }
    }
  }
}

closed 2020-06-01 by Snowist