pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #4: HEX for ESP32 upload

Dfinitski, 2020-11-17:
Hi. I am not able to compile this project. Where can I find ready to upload HEX file ?
It would be perfect keep releases here.

pe0mgb, 2020-11-17:
David,Sorry, I have no idea how to make a hex file for the ESP32. I will try to figure it  out.Also to upload I have to check.Regards Gert PE0MGBVerzonden vanaf Samsung-tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/4",
"url": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/4",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Dfinitski, 2020-11-17:
Hi. Try Sketch -> Export compiled binary

leogoto, 2020-11-17:
Hi David,

Here is hex file.

Leo/WH2Q

Dfinitski, 2020-11-18:
Hi, Leo.
Where is exactly ? Please give a link.

leogoto, 2020-11-18:
Hi David,

I do compiling.

Dfinitski, 2020-11-18:
Yes, "Export compiled binary" option creates bin file in the same directory where placed ino file.
Later everyone could upload that file into ESP32 without compilation of the project.
Sorry, initially I thought hex files format is working binary file for Arduino.
No, bin file of course.

closed 2020-11-20 by Dfinitski