pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #5: Compilation error

Dfinitski, 2020-11-18:
Hi.
I just installed latest TFT_eSPI-master and SI4735-master,

C:\SI4735-Radio-ESP32-Touchscreen-Arduino-master\si4735_2.8_TFT\si4735_2.8_TFT.ino: In function 'void setup()':
si4735_2.8_TFT:769:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calData);
         ^
C:\SI4735-Radio-ESP32-Touchscreen-Arduino-master\si4735_2.8_TFT\si4735_2.8_TFT.ino: In function 'void loop()':
si4735_2.8_TFT:1103:19: error: 'class TFT_eSPI' has no member named 'getTouch'
     pressed = tft.getTouch(&x, &y);
                   ^
C:\SI4735-Radio-ESP32-Touchscreen-Arduino-master\si4735_2.8_TFT\si4735_2.8_TFT.ino: In function 'void subrstatus()':
si4735_2.8_TFT:2296:18: error: 'class TFT_eSPI' has no member named 'getTouch'
     press1 = tft.getTouch(&x, &y);  
                  ^
exit status 1
'class TFT_eSPI' has no member named 'setTouch'

pe0mgb, 2020-11-18:
Read my github , how to use tft_espiVerzonden vanaf Samsung-tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/5",
"url": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/5",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Dfinitski, 2020-11-18:
Thanks for the update.
Now compilation is successful.

closed 2020-11-18 by Dfinitski