pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #6: 2.8" TFT SPI 240*320

Haranochi, 2021-01-03:
I have finished building your project and it works correctly, the problem I have is that the touch screen does not work, I use a Chinese display with reference KMRTM28028-SPI, (ILI9341), I cannot find your data sheet on the Internet, can you help me to solve the problem. Thank you.

https://user-images.githubusercontent.com/76921787/103490211-8a386500-4e1a-11eb-8871-fd9760534a44.MOV

pe0mgb, 2021-01-03:
I'm afraid it is not a touchscreen. Please send me a picture of the backsite.Verzonden vanaf Samsung-tablet.

https://user-images.githubusercontent.com/76921787/103490211-8a386500-4e1a-11eb-8871-fd9760534a44.MOV

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/6",
"url": "https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/6",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Haranochi, 2021-01-04:
Thanks for your answer, I attach an image of the back of the display, you will see that it has an IC (HR2046), I suppose a clone of the XPT2046, ¿maybe it be defective?
In the preamble of your sketch it indicates that the display is powered by 3V3, but in your scheme the VCC of the display indicates 5V, of course it can be powered by 3V3 or 5V, your display with what voltage is it powered, in my construction I feed it with 5V If I had to feed it to 3V3, I would have to make a jumper on J1 of the display and change the supply on the pcb to 3V3. I see in the file Setup1_ILI93441.h the clock frequency of the XPT2046 is 25MHz, can it be modified to a value of 20MHz ?. The balun you use what type of copper wire you use. Thank you for your response and for making your project available to millions of people.
[image]

pe0mgb, 2021-01-04:
Hello Haranochi.

Thanks for the picture. What I have found on the internet that the chips are the same. Same functionality. An other story was that from your displays touch screen has different dimensions and is reverse mounted. I can not check all this information. We all know that there enough nonsense available on the Web.

Anyway I don’t know but also the layout of the TFT pcb is different as mine.  I was even not aware that other different version of the display exist.

 

 <https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

This is where I have ordered my displays and a lot of other people did. Order the 2.8 TFT SPI with touch 240*320. This is with the ILI9341 and the XPT2046.

I have V1.2. Hope this will solve your problem. This Aliexpress supplier is normally very fast. !0 days here in the Netherlands.

 

Thanks for your remark about the 3.3 v and 5v supply of the display. I think it is better to use the 5v for the display. Then the regulator at the display is now making the 3.3v and is generating some heath . 

When you supply it with 3.3v (J1)  then the esp32 regulator has to do all the work. I prefer the first proposal. I’m using 5v.

Your 25MHz or 20 MHz I don’t know how it’s working, just try it.

I’m not sure about your balun question. Are you talking about the transformer with 50 and 10 turns?. Well with the FT50-43 I use 0,2 mm or 0,25 mm. 0.3 is possible but then you may have a some spacing problem.

 

Best Regards,

 

Gert PE0MGB

 

 <https://user-images.githubusercontent.com/76921787/103532779-23539400-4e8c-11eb-91e4-979a438f5ba2.jpg> Afbeelding verwijderd door afzender. IMG-4069

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

Haranochi, 2021-01-04:
Hi Gert,

 

Thanks again, another question in the Setup1_ILI9341.h file, has the following definition;

 

// For ESP32 Dev board (only tested with ILI9341 display)

// The hardware SPI can be mapped to any pins

 

#define TFT_MISO 19

#define TFT_MOSI 23

#define TFT_SCLK 18

#define TFT_CS 15 // Chip select control pin

#define TFT_DC 2 // Data Command control pin

#define TFT_RST 4 // Reset pin (could connect to RST pin)

#define TOUCH_CS 5 // Chip select pin (T_CS) of touch screen

 

#define TFT_MISO 19, does not connect, but Touch_DO (SPI bus out) also uses pin 19, are there problems with this configuration?

 

Best Regards

 

José María

Enviado el: lunes, 04 de enero de 2021 15:48
Para: pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino <SI4735-Radio-ESP32-Touchscreen-Arduino@noreply.github.com>
CC: Haranochi <haranochi@telefonica.net>; Author <author@noreply.github.com>

 

<https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

https://www.avast.com/antivirus

pe0mgb, 2021-01-04:
Hello José,

 

Good question about D19. Pin9 or SDO is the data-out of the TFT and has to be used if you want to make a dump of a certain screen.. To save pins some screen and touch pins are connected in parallel. 

The selection is done with CS (chip select) pin 3 for the TFT and T_CS (pin 11) for the touch screen. Everything is working fine with the SDO(pin9) and T_DO(pin13) in parallel. I’m also working at a 3.5 inch display that you can control  with the same hardware configuration. However the SDO (pin9) at 3.5 TFT is not tri-state. and keep pin 13(T_DO) fixed and the touch is not working. Because SDO (pin9) is not used in our application I do not connect it anymore.  Now the 2.8 and 3.5 inch TFT display are working in the same hardware configuration and can be interchanged. Be care full you still need dedicated software to control the different displays. Hope this answered your question. I’m sure this has nothing to do with your problem.

 

Regards,

Gert PE0MGB

 

<https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

https://www.avast.com/antivirus

Haranochi, 2021-01-04:
Gert, good night

 

No, it is not related to my problem, curiosity to know the use of D19, thanks for the explanation, researching I have seen that pin 14 (T_IRQ) is always at high level and when it touched any menu on the screen it falls to low, this is correct, it detects a pressed menu key, now my question is about pin 11 (T_CS), it is high or low, in my case it is always high, it does not change, and I think it must be low to enable chip select signal. If you have time, you can explain to me the operation of the T_IRQ and T_CS pins. The Chinese display works with the Touch_calibrate example from the TFT_eSPI library, connecting it as indicated by the creator of the library, bodmer.

 

Regards

 

José María

Enviado el: lunes, 04 de enero de 2021 19:58

 

<https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

https://www.avast.com/antivirus

pe0mgb, 2021-01-05:
José,

I understand that the calibration program is running and you can calibrate the TFT. That means that everything works. You can run the program on the radio without any change in files. Just download the calibration-program. Don’t forget the rotation. When the program works try to replace the calibration data in my program with the generated data by yourself. You can find this information at the beginning of the setup routine. When the program is not working, please check the display wiring.

Please let me know about the results.

 

Gert

 

<https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

https://www.avast.com/antivirus

Haranochi, 2021-01-06:
Hi Gert, 

 

Thank you for your patience to help me, if I insist on writing to you, it is because I am going to the University soon and I want to take with me the design of your radio, I have carried out the following tests and these are the results;

 

The wiring for the TFT calibration is carried out as indicated in its diagram, which is as follows:

  

Display                                                 ESP32

 

  VCC                                                    +5V

  GND                                                   GND

  CS                                                       D15

  RST                                                     D4

  DC                                                       D2

  SDI(MOSI)                                         D23

  SCLK                                                   D18

  LED                                                   +5V

  SDO(MISO)                                       NC

  T_CLK                                                D18

  T_CS                                                  D5

  T_DIN                                                D23

  T_DO                                                 D19

  T_IRQ                                                D34

 

  I have checked the display connections, testing continuity with a multimeter from the display connector to the ESP32 and they are correct, I use a 30-pin ESP32 WROOM Devkit.

  The wiring to calibrate the screen I have done as your schematic and it works.

  I have placed the results of the display calibration in the lines of the sketch, 776 and 782, the display does not respond to communicate with the sketch, I observe that T_IRQ when I touch a button on the first screen, at the moment I cannot go to the next, changes its value from 3.54V to 0.0V, but the sketch does not recognize this change, I understand that T_IRQ when it changes value indicates that the touch screen works.

 

  TFT 2.8" Calibration

 

Rotation Horizontal = 1  (Calculated, Touch_Calibrate, TFT_eSPI)

// Use this calibration code in setup():

uint16_t calData[5] = { 503, 3371, 311, 3345, 7 };

tft.setTouch(calData)

========================================================

#ifdef IhaveHoriTFT  // Your software; Lines of his sketch 774 

 // Calibration code for touchscreen : Rotation = 1

uint16_t calData[5] = { 387, 3530, 246, 3555, 7 };

tft.setTouch(calData);

========================================================

 

Rotation Vertical= 2  (Calculated, Touch_Calibrate, TFT_eSPI)

// Use this calibration code in setup():

uint16_t calData[5] = { 403, 3328, 548, 3372, 2 };

tft.setTouch(calData)

========================================================

#ifdef IhaveVertTFT // Your software; Lines of his sketch 780

// Calibration code for touchscreen : for 2.8 inch & Rotation = 2

uint16_t calData[5] = { 258, 3566, 413, 3512, 2 };

tft.setTouch(calData);

 

I have looked at your sketch to know how it recognizes when a key on the screen has been pressed and I get to:

 

//=======================================================   Buttons First and Third Layer   ==========================

typedef struct // Buttons first layer

{

  const char *ButtonNam;

  

  int    ButtonNum;       // Button location at display from 0 to 11. To move around buttons freely at first layer.

  const char *ButtonNam1;

  int    ButtonNum1;      // Button location at display from 0 to 11. To move around buttons freely at third layer.

  int    XButos;          // Xoffset

  long   YButos;          // Yoffset

} Button;

 

int ytotoffset = 0;

 

//  Button table

int Xbutst  =   0;               // X Start location Buttons

int Ybutst  = 160 + ytotoffset;  // Y Start location Buttons

 

int Xsmtr   =   0;

int Ysmtr   =  85 + ytotoffset;  // S meter

 

int XVolInd =   0;

int YVolInd = 135 + ytotoffset;  // Volume indicator

 

int XFreqDispl  =   0;

int YFreqDispl  =   0 + ytotoffset;  // display

 

int Xbutsiz =  80;  //size of buttons first & third layer

int Ybutsiz =  40;

 

#ifdef IhaveVertTFT

int Xbut0  = 0 * Xbutsiz ; int Ybut0  = 0 * Ybutsiz; // location calqualation for 12 first layer buttons

int Xbut1  = 1 * Xbutsiz ; int Ybut1  = 0 * Ybutsiz;

int Xbut2  = 2 * Xbutsiz ; int Ybut2  = 0 * Ybutsiz;

int Xbut3  = 0 * Xbutsiz ; int Ybut3  = 1 * Ybutsiz;

int Xbut4  = 1 * Xbutsiz ; int Ybut4  = 1 * Ybutsiz;

int Xbut5  = 2 * Xbutsiz ; int Ybut5  = 1 * Ybutsiz;

int Xbut6  = 0 * Xbutsiz ; int Ybut6  = 2 * Ybutsiz;

int Xbut7  = 1 * Xbutsiz ; int Ybut7  = 2 * Ybutsiz;

int Xbut8  = 2 * Xbutsiz ; int Ybut8  = 2 * Ybutsiz;

int Xbut9  = 0 * Xbutsiz ; int Ybut9  = 3 * Ybutsiz;

int Xbut10 = 1 * Xbutsiz ; int Ybut10 = 3 * Ybutsiz;

int Xbut11 = 2 * Xbutsiz ; int Ybut11 = 3 * Ybutsiz;

#endif

 

¿What function does Button; ¿(creation of virtual switches on display?). 

 

This line, 2303 --- press1 = tft.getTouch(&x, &y); recognizes when a virtual switch on the screen is pressed.

 

¿(& x, & y), are the coordinates sent by the display to sketch, where they are saved?.  I'm sure this summer I will do a C ++ programming course. The 3V3 are not supplied by the ESP32, they are supplied by a voltage regulator LM2937ET, maximum current supported 500mA, the voltage measured at its output 3.27V. Also note that T_CS is high on pin D5, but the XPT2046, the CS signal is negated (NOT) from what I understand that the touch panel is enabled. Sorry for being insistent, in Spanish it is said "muy cansino"

 

Regards

 

José Maria

Enviado el: martes, 05 de enero de 2021 21:06

 

<https://a.aliexpress.com/_B1zwoE> https://a.aliexpress.com/_B1zwoE

https://www.avast.com/antivirus

Kuddl5, 2021-08-05:
Hallo,
Ich kann den ESP32 nicht installieren. Könnte mir jemand helfen eine Quelle zu finden um 5 ESP32 ,programmiert mit
Pe0mgb Programm TFT 2,8 si4732 radio, zu Kaufen?

pe0mgb, 2024-04-08:
> Hi Gert. I have also built from one of your PCB's, all ok ok apart from the touch screen. It does work but the layout is mirrored. All the icons on the bottom of the  screens appear at the bottom of the display but only work if you press in the top half of the display - for example the 'NEXT' button only works if you press in the top left hand corner and the 'MODE' only works if you press in the top right hand corner, to open the 'HAM' screen you have to press above the 'freq' button on the bargraph.... It is all mirrored  I have looked for a resolution without success so far. My display is a V1.1 - like the one you link above, but I see your actual display was a V1.2 and wonder if that is the issue. It is just possible that as these things are assembled 'cheap as chips' I might just have a bad one.  I am going to try ordering a V1.2 and hope that fixes it.
> Many thanks for all your hard work.
> Mike

Hello Mike,I think nothing wrong with the radio but you have bad luck with the display. You have a rotated touch screen. For about 4 years the display was always the same and no problem. All of a sudden the last months you get from some suppliers a rotated touch screen. We can easy solve that with the software. Which version are you using?Regards,Gert PE0MGB

dart1280, 2024-04-09:
Hi Gert, Oh...ok!   I had a troll through the firmware looking for
something that looked like it might be able to rotate the touch screen 180
degrees and while a couple of places looked promising I wasn't successful.
I also tried entering the calibration values obtained from running the
calibration routine but that stopped the touch working totally.  I loaded -

SI473X_2.8_TFT_V3.5.  04-03-2022
Part Number 23
Major rev. 54
Minor rev. 48
Patch ID 00
component Major rev. 55
component minor rev. 48
Chip revision 68

Thanks for your help.

Regards

Mike

dart1280, 2024-04-09:
Well...there's a thing!    After editing the Touch_Calibrate sketch to
allow the screen backlight to come on during the calibrate test
(digitalWrite(14, LOW); ) I could now see the screen , before I couldn't
see anything on the screen so wasn't pushing in the right places!  I
successfully completed the calibrate test and after inserting those
calibrated values into your firmware ( line 983 for the vertical screen    {
340, 3492, 195, 3494, 4 };) the touch now works correctly.

I guess I should do the calibrate test for the other screen orientations
too huh!

So thankyou for your interest and I think the problem is sorted.

Regards

Mike

z34pec, 2024-09-06:
Hello. 
As a addition to this conversation, how is posible to use 2.4" TFT SPI 240*320 screen with your sketch.
I am not so close with arduino and ESP32 programming but I can make some changes in codes following instructions.
Here is picture of my display:
[image]