pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #7: Touch calibration

codeput, 2021-04-26:
I have problems with entering frequencies directly, the touch is not perfectly centered, is it possible to draw the keyboard slightly to the right?

pe0mgb, 2021-04-26:
Hello,

Normally with a 2.8 inch display I have never seen a problem. I think you are using a screen with another size. What will help is calibrating the screen. 

In the TFT_eSPI library in examples/Generic there is a calibration program called Touch calibrate. Here you can calibrate the display. Think about the rotation !!! This is very important.

The found figures you have to fill in in  the setup part of my sketch. See line 774 and 780

I have never checked the sketch with an other display . But I think it is possible the give an offset. I will check it the coming days. But it will work only for the main screen.

Let me know if it works or not..

Regards,

Gert PE0MGB

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

codeput, 2021-04-26:
You are right, my display is 2.4 inches so it is not possible to correct
the position of the touch?

pe0mgb, 2021-04-26:
No, the program is written for 2.8 inch. With 2.4 of course it is posible. You only have to rewrite a part of the programma. Forget it. Buy a 2.8 and your problemen are solved. 

⁣BlueMail voor Android downloaden ​

pe0mgb, 2021-04-27:
When your 2.4 tft had 240*320 pickels Just calibrate and it will work the same as a 2.8

⁣BlueMail voor Android downloaden ​

codeput, 2021-04-27:
Thanks a lot Gert, I solved it using the bookmark utility in the
library. Now it works perfectly.

codeput, 2021-04-27:
Thanks a lot Gert, I solved it using the calibration utility in the
library. Now it works perfectly.

yo2ldk, 2021-04-27:
Hi Gert,
  when I  touch the broad band, input freq I can see that window, then my display frozen and need to reset
  work fine in FM start.
  please any tips / advice
  
   best regards,
   
      de yo2ldk - alexander

pe0mgb, 2021-04-27:
Hello Alexander,
It looks like the program stopped. No Idea what happen. I have no experience with the Wemos device. It looks like the problem is there. I use always the 30 pin ESP32. I realize this is a bad answer but I really don't know and I can not simulate the problem. If posible try a 30 pin ESP32 and it will work.

Regards,
Gert PE0MGB

⁣BlueMail voor Android downloaden ​

yo2ldk, 2021-04-27:
Thank you Gert a lot, for your answer and  your version for SI4735.
the rotary knob also running in a very strage way some time good, sometime after 3-5 rotations,
so, I change the pinouts of the encoder to D25 and D25, sw to D39, and now the radio is ALIVE!  :)
 I cannot explain what it was, but I'm happy with him running..
 
   all the best and 73,
   
    yo2ldk -alexander
    TNX AGN !

closed 2021-04-29 by codeput