pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #11: display does not dim or turn off after 5min

larrykabo, 2021-11-18:
This is stated in v 3.3 release notes

pe0mgb, 2021-11-18:
Are you using an ATS25?
BlueMail <https://bluemail.me> for Desktopdownloaden
larrykabo wrote:
This is stated in v 3.3 release notes

larrykabo, 2021-11-18:
yes, ats25

pe0mgb, 2021-11-18:
Larry,
The problem is that the ATS-25 is for 99% a copy of my hardware and software design. All my software is developed for my developed hardware and is copied by some people . To the best of my knowledge there are in the ATS25 design 3 transistors missing. No idea why. Maybe cheaper. One of the transistors is able to switch off or dim the backlight of the TFT. That works in my hardware. Unfortunately in the ATS25 the transistor is missing. The other two transistors are used to switch off the LF during bandswitching because of a very nasty click in LF. In the ATS25 it is loud present.
I have never build a SI5351 in an ATS25 .Way to difficult with all the small tracks and SMD components. I'm still trying to get the important software parts up and running in the ATS25 but don't blame me when it is not working. I have spend a lot of time with testing the ATS25. With a good antenne it works very well. Do not expect miracles from a rod-antenna at the backsite. First of all there is some distorsion from the radio. ESP32 controller and the TFT screen are causing a lot of noise. I have found that the aluminum box is not grounded. You can do that simple with a connection with a piece of wire from the ground of the antenna connector and two bolt connected to the upper and lower part of the housing. This reduced the noise already at lot but it is not gone.
An wire antenne, I prefer a dipole, give you an excelent signal from many station. I'm still impressed about the performance of SI4732 or SI4735 radio chip using the library developed by Ricardo PU2CLR. Without that library my radio didn't excist. Hope that this clears a little bit the situation.
Regards,
Gert PE0MGB
BlueMail <https://bluemail.me> for Desktopdownloaden
larrykabo wrote:
yes, ats25

larrykabo, 2021-11-19:
Thanks for your patience and great explanation, do you know what the
missing chip on the back of display board might be for?

Larry

larrykabo, 2021-11-19:
Also, the two halves of the ATS25 will not ground together, that anodized
aluminum needs some grinding/sanding to complete the whole shell ground,
but stuffs going to leak in the front and back.