pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #9: changes for clock schematic is not on github

larrykabo, 2021-11-18:
Not only that it defaults to the hardware modified version, most people like me took time to figure this out. 2nd where are the hardware things I need to do to use the improved option?

pe0mgb, 2021-11-18:
I'm not sure if I understand your question. But how to connect the SI5351 is in my schematics. When stay at the crystal you don't need any additional hardware. For the ATS-25 as I told you, only the display dimmer is not working due to a missing
transistor.
⁣BlueMail voor Android downloaden ​

larrykabo, 2021-11-18:
is it possible for you to provide pics of the si5351 mods? the
schematics included with that release show jumpers etc of certain ones need
to be changed, no jumpers on the board to be moved and the schematic does
not show all the chip in the unit across the 2 boards yet there are chip in
the schematic that are located on 2 different boards, like the usb
interface is nowhere in the schematics. is there another full schematic?.
On the front board there is a missing chip, is this where the si5351 goes?