pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #14: AGC enables in SSB after tuning so far

madmax2069, 2022-01-29:
I'm using an ATS-25 running 3.4

When in SSB and I manually turn AGC off when you tune so far (like 15KHz) you'll hear a chuff sound and the AGC will be enabled but the button isn't highlighted like it's off.

When you press the AGC button it'll turn the AGC back off and the button will stay un-highlighted to match the AGC being off. But you tune so far again it'll make that chuff sound and the AGC will be enabled itself again with the button showing that it's off.

pe0mgb, 2022-01-30:
Hi MadMax,

Thanks for all the comments.

1) The medium wave. That's a problem with all the different band plans in the world. It's not just on medium wave. There are also worldwide differences in the Hambands. I will create a separate band format for the North American region that you can select with a #define to run the compiler. To ease the suffering now you can easily make a change for the MW manually.

In version 3.4 you will find in the following line:

 

661 { "MW", MW_BAND_TYPE, AM, 522, 1701, 1395, 9, 0, 0}, // MW 2

 

Change it as below. This is the North American band plan. 1390 means 1390 KHz and is a preferred frequency that the program starts at the first time.

 

661 { "MW", MW_BAND_TYPE, AM, 530, 1700, 1390, 10, 0, 0}, // MW 2

 

2) Sorry but I don't understand this question.

3) The next comment was about the behavior of the radio with the BFO frequency and the secretive changing of the frequency. You are absolutely right. That's a really nasty bug. This will be fixed in my next release.

4) Your comment about the AGC turning back on with a frequency change is a known old problem. The cause here is in the software patch that is always loaded for SSB. No source is available, but it does contain SSB's secret for this chip. You may be able to create a bypass in the library, but that has not been implemented so far. I'll pay attention to it again. One consolation is that SSB works best with the AGC on.

I'll keep you posted on the progress.

Best Regards,

Gert PE0MGB

madmax2069, 2022-01-31:
Hello, thanks for the quick reply

 I just tried making the edit for the MW/AM frequencies and during compile process I get an error

Fatal error: Rotary.h: no such file or directory, so I take it I'm missing a library ? If I am what library do I need to install ?

pe0mgb, 2022-01-31:
You can find the library at my github
Outlook voor Android downloaden

madmax2069, 2022-02-01:
I figured it out (pulled the files I needed from the .zip I downloaded from here), made the change to the MW frequency range and pushed the change to the radio. The only thing I touched in the sketch is what you told me to change for the MW frequency range.

But now AGC isn't working correctly like how it use to work before the change, AGC isn't recovering when changing the frequency. 

When I turn the radio on now (AGC is on by default which is normal) but regardless of where I had the volume set (I could have the volume set to 50) the sound is really low (like the volume is around 5). When I disable AGC the the sound loudness returns to normal. And when I enable the AGC after turning AGC off the sound is just as loud as with AGC off. BUT as soon as I turn the channel/Frequency with AGC on the sound loudness drops really low again and doesn't recover like it use to do, before when AGC was enabled and I changed the frequency there would be a temporary drop in sound loudness and would slowly recover back to a normal loudness. 

I'm probably going to have to make a video showing you what I'm talking about as I'm not the best explaining things.

madmax2069, 2022-02-01:
Here's a video of what I'm talking about. https://youtu.be/nIRpz8nPmw0