pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #15: compiling V3.0-3.4 with IDE1.8.15 not succesfull

hamroel, 2022-02-20:
compiling version 3.0 - 3.4  gives a bunch of errors for all versions:

in IDE opened the ino + rotary.cpp + rotary.h + TFT_eSPI.cpp + TFT_eSPI.h + patch_init.h

1)
SI4735_2.8_TFT_V3.0:732:16: error: expected initializer before 'RotaryEncFreq'
 void IRAM_ATTR RotaryEncFreq()  {
                ^~~~~~~~~~~~~
SI4735_2.8_TFT_V3.0:732:16: error: expected initializer before 'RotaryEncFreq'
 void IRAM_ATTR RotaryEncFreq()  {
                ^~~~~~~~~~~~~

code =  
void IRAM_ATTR RotaryEncFreq()  {
...
}
removing IRAM_ATTR solves this error.
what means IRAM_ATTR?

2)
path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino: In function 'void setup()':
SI4735_2.8_TFT_V3.0:781:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calData);
         ^~~~~~~~

the requested library got from https://github.com/Bodmer/TFT_eSPI + imported it...

3)

SI4735_2.8_TFT_V3.0:790:32: error: no matching function for call to 'EEPROMClass::begin(int)'
   if (!EEPROM.begin(EEPROM_SIZE))
                                ^
In file included from path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino:68:0:
C:\Users\nn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\EEPROM\src/EEPROM.h:125:11: note: candidate: EEPtr EEPROMClass::begin()
     EEPtr begin()                        { return 0x00; }
           ^~~~~
C:\Users\nn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\EEPROM\src/EEPROM.h:125:11: note:   candidate expects 0 arguments, 1 provided
SI4735_2.8_TFT_V3.0:815:57: error: 'RotaryEncFreq' was not declared in this scope
   attachInterrupt(digitalPinToInterrupt(ENCODER_PIN_A), RotaryEncFreq, CHANGE);
                                                         ^~~~~~~~~~~~~
SI4735_2.8_TFT_V3.0:899:3: error: 'xTaskCreate' was not declared in this scope
   xTaskCreate(SaveInEeprom, "SaveInEeprom", 2048, NULL, 1, NULL);
   ^~~~~~~~~~~
path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino: In function 'void SaveInEeprom(void*)':
SI4735_2.8_TFT_V3.0:928:12: error: 'struct EEPROMClass' has no member named 'commit'
     EEPROM.commit();
            ^~~~~~
SI4735_2.8_TFT_V3.0:930:23: error: 'portTICK_RATE_MS' was not declared in this scope
     vTaskDelay(5000 / portTICK_RATE_MS);
                       ^~~~~~~~~~~~~~~~
SI4735_2.8_TFT_V3.0:930:5: error: 'vTaskDelay' was not declared in this scope
     vTaskDelay(5000 / portTICK_RATE_MS);
     ^~~~~~~~~~
path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino: In function 'void saveConfig()':
SI4735_2.8_TFT_V3.0:943:10: error: 'struct EEPROMClass' has no member named 'commit'
   EEPROM.commit();
          ^~~~~~
path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino: In function 'void loop()':
SI4735_2.8_TFT_V3.0:1158:19: error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
     pressed = tft.getTouch(&x, &y);
                   ^~~~~~~~
                   getColor
path\SI4735_2.8_TFT_V3.0\SI4735_2.8_TFT_V3.0.ino: In function 'void subrstatus()':
SI4735_2.8_TFT_V3.0:2315:20: error: 'class TFT_eSPI' has no member named 'getTouch'; did you mean 'getColor'?
     presStat = tft.getTouch(&x, &y);
                    ^~~~~~~~
                    getColor
Meerdere bibliotheken gevonden voor "SI4735.h"
Gebruikt: C:\Users\nn\Documents\Arduino\libraries\SI4735-master
Niet gebruikt: C:\Users\nn\Documents\Arduino\libraries\PU2CLR_SI4735
exit status 1
expected initializer before 'RotaryEncFreq'

the errors are a bit complicated so addressed as an issue.

and also is it possible to compile it and provide the bin file to load with esptool.exe?

regards Roel

pe0mgb, 2022-02-22:
Hi Roel,
You have a big problem with the libraries. Installed the ESP32 from Espressif systems in the Arduino board manager?  Then the ESP32 Dev. Module selected? In my github you can read in Readme what to do with the configuration files for the library TFT_eSPI. The TFT_eSPI library just needs to be installed in the library directory under the Arduino directory. I don't make bin files if only that some program options have to be selected in the ino.
Good luck,
Gert PE0MGB

hamroel, 2022-02-22:
the                error: 'class TFT_eSPI' has no member named 'setTouch' tft.setTouch(calData);                      has been gone:

-the directory with the ino-file contains:
patch_full.h
patch_init.h
Rotary.cpp
Rotary.h
SI4735_2.8_TFT_V3.0.ino

-the (latest) library TFT_eSPI-master.zip has been added to the libraries

-the "User_Setup_Select.h" provided with the versionpackage placed in
path\Arduino\libraries\TFT_eSPI-master

       note the path of the include in "User_Setup_Select.h" is correct:
         #include <User_Setups/Setup1_ILI9341.h>  // Setup file configured for my ILI9341  2.8 inch

-the "Setup1_ILI9341.h" provided with the versionpackage placed in
path\Arduino\libraries\TFT_eSPI-master\User_Setups overwriting the "Setup1_ILI9341.h" of the library

Roel

hamroel, 2022-02-22:
SI4735_2.8_TFT_V3.0:790:32: error: no matching function for call to 'EEPROMClass::begin(int)'
if (!EEPROM.begin(EEPROM_SIZE))

has been solved: 

somehow I selected before indeed the ESP32 Dev Module at first but the LDE has selected later on the previous Anet board.

succesfully compiling is possible:

De schets gebruikt 336354 bytes (25%)  programma-opslagruimte. Maximum is 1310720 bytes.
Globale variabelen gebruiken 18120 bytes (5%) van het dynamisch geheugen. Resteren 309560 bytes voor lokale variabelen. Maximum is 327680 bytes.

closed 2022-02-22 by hamroel