pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #16: BIN(hex) files for ESP32 and upload tool V3.0-V3.4

hamroel, 2022-02-27:
If somebody is able to compile  an INO-file with its libraries the whole compiling-history is stored in a local temp-directory
C:\Users\nn\AppData\Local\Temp\arduino_build_400xxx
There appear also the bin files for ESP32 and when copied before IDE is closed you can use them for ESPtool.exe
For SI4735_2.8_TFT_V3.4 it will be like this:

26-03-2021  12:26            17.120 bootloader_dio_80m.bin
26-03-2021  12:26             8.192 boot_app0.bin
12-04-2021  22:22             3.519.420 esptool.exe
27-02-2022  11:38            560 flash.bat
27-02-2022  10:57            407.520 SI4735_2.8_TFT_V3.4.ino.bin
27-02-2022  10:57            3.072 SI4735_2.8_TFT_V3.4.ino.partitions.bin

If running the flash.bat (modified for SI4735_2.8_TFT_V3.4) it skips the whole compiling process and pumps it into the ESP32 WROOM

content of flash.bat:

@echo off
echo -----------------------------
echo - Update tool ESP32 SI4735  -
echo -       Version V3.4        -
echo -----------------------------
echo.
echo Available COM-ports:
chgport
echo.
set /p COM=Enter Comport number (example: 3):
esptool.exe --chip esp32 --port COM%COM% --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 boot_app0.bin 0x1000 bootloader_dio_80m.bin 0x10000 SI4735_2.8_TFT_V3.4.ino.bin  0x8000 SI4735_2.8_TFT_V3.4.ino.partitions.bin 
@pause

regards Roel

hamroel, 2022-02-27:
[SI4735_2.8_TFT_V3.4 gecompileerd.zip](https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/files/8148969/SI4735_2.8_TFT_V3.4.gecompileerd.zip)
[supporting files.zip](https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/files/8148972/supporting.files.zip)

pe0mgb, 2022-03-25:
With the zip file you can not make your own selections in the different options.