pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #18: Rotary Encoder Reversed in 3.5

smannchen, 2022-04-01:
I just purchased my ATS25 and it shipped with Firmware 3.1 after upgrading to 3.5 I noticed the tuning was backwards.

To resolve this issue I changed line 946 from CW to CCW and it works. 

Originally
946      if (encoderStatus == DIR_CW)// Direction clockwise 

Changed to
946      if (encoderStatus == DIR_CCW)// Direction counter-clockwise

jalobaugh, 2022-04-03:
Thanks for this. I've just purchased an ATS-25 and found out there was an update that included the N/S American band plans. I'm new to Arduino and this threw me for a loop. I figured this was the line to change, but wasn't sure on the syntax.

smannchen, 2022-04-03:
Glad I could help. Was my first time playing with this radio, and I was confused why it was tuning backwards after the upgrade lol.