pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #21: Compile Error

ESSCHENK1, 2022-06-05:
Hello,

I have an error while compiling version3.5
version 3.4 is working nice.

There whas no adc.h file in the zip.
so i found something online and use that.

Thanks in advanced.
Ed PA1TR

This is the error i get.

Arduino:1.8.13 (Windows 10), Board:"ESP32-WROOM-DA Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

C:\arduino\examples\SI473X_2.8_TFT_V3.5\SI4735_2.8_TFT_SI5351_V3.5\SI4735_2.8_TFT_SI5351_V3.5.ino: In function 'uint32_t readADC_Cal(int)':
SI4735_2.8_TFT_SI5351_V3.5:3440:3: error: 'esp_adc_cal_characteristics_t' was not declared in this scope
   esp_adc_cal_characteristics_t adc_chars;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
SI4735_2.8_TFT_SI5351_V3.5:3442:28: error: 'ADC_UNIT_1' was not declared in this scope
   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                            ^~~~~~~~~~

SI4735_2.8_TFT_SI5351_V3.5:3442:40: error: 'ADC_ATTEN_DB_11' was not declared in this scope

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                                        ^~~~~~~~~~~~~~~

C:\arduino\examples\SI473X_2.8_TFT_V3.5\SI4735_2.8_TFT_SI5351_V3.5\SI4735_2.8_TFT_SI5351_V3.5.ino:3442:40: note: suggested alternative: 'ADC_ATTENDB_MAX'

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                                        ^~~~~~~~~~~~~~~

                                        ADC_ATTENDB_MAX

SI4735_2.8_TFT_SI5351_V3.5:3442:57: error: 'ADC_WIDTH_BIT_12' was not declared in this scope

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                                                         ^~~~~~~~~~~~~~~~

SI4735_2.8_TFT_SI5351_V3.5:3442:82: error: 'adc_chars' was not declared in this scope

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                                                                                  ^~~~~~~~~

C:\arduino\examples\SI473X_2.8_TFT_V3.5\SI4735_2.8_TFT_SI5351_V3.5\SI4735_2.8_TFT_SI5351_V3.5.ino:3442:82: note: suggested alternative: 'u_char'

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

                                                                                  ^~~~~~~~~

                                                                                  u_char

SI4735_2.8_TFT_SI5351_V3.5:3442:3: error: 'esp_adc_cal_characterize' was not declared in this scope

   esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);

   ^~~~~~~~~~~~~~~~~~~~~~~~

SI4735_2.8_TFT_SI5351_V3.5:3443:11: error: 'esp_adc_cal_raw_to_voltage' was not declared in this scope

   return (esp_adc_cal_raw_to_voltage(ADC_Raw, &adc_chars));

           ^~~~~~~~~~~~~~~~~~~~~~~~~~

pe0mgb, 2022-06-05:
This is a part of the ESP32 in board manager. Install the latest version and the problem is solved.Outlook voor Android downloaden

ESSCHENK1, 2022-06-06:
Thanks Gert its working perfect now.
Thanks for al the nice work.
73 Ed