pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #23: TFT rotate

FabioLucca01, 2023-03-17:
I assembled the standard circuit, the same as the project (ILI9341 + ESP32WROOM32 + Si4735), however the display is not in the correct rotation, both vertically and horizontally it does not draw all the buttons, and the position of the buttons is always incorrect in relation to screen, I already messed with the configuration in many points of the code without success. I'm using the updated libraries, is there a problem related to the new library? Has anyone ever experienced this?
[image]

pe0mgb, 2023-03-17:
Hello Fabio,It is a pitty that you are facing some problems. Let me try to help you.What is the size of the screen? Can you send me a picture of the screen with the original sketch?Greetings.Gert PE0MGB

FabioLucca01, 2023-03-17:
> 

Hello, thanks for replying. Screen is standard 2.8" 320x240 ILI9341. I used version 3.5 of the project as per repository and replaced the 2 configuration files in the TFT library.

pe0mgb, 2023-03-19:
Fabio,

I just have checked V3.5 with the latest version of TFT_eSPI and I didn’t find any problem.

When I look at the picture of the screen you have send, colors and size are wrong. This is related to a problem in configuration of the library software. Not related with the V3.5 radio ino. 

Can you send me a picture of the backside of the TFT screen?

Regards,

Gert PE0MGB

pe0mgb, 2023-03-19:
//#include <Setup3_ILI9488.h>       // Setup file configured for my ILI9488  3.5 inch

#include <Setup1_ILI9341.h>         // Setup file configured for my ILI9341  2.8 inch  esp32       <=============================================

//#include <Setup4_ILI9341_8266.h>  // Setup file configured for my ILI9341  2.8 inch  8266

//#include <Setup2_ST7735.h>        // Setup file configured for my ST7735

 

Is line 27 in the User_Setup_Select file like above

FabioLucca01, 2023-03-19:
Hi, my "User_Setup_Select" is correct like yours above:
#include <Setup1_ILI9341.h> // Setup file configured for my ILI9341 2.8 inch esp32

The ".ino" sketck is correct, I used a standard example from the TFT_eSPI library and the display doesn't work right, always cutting off some part of the screen, I will study the library configuration files and make some changes to see if it fixes it, if I find it the solution I return here. Thanks for the help.
[image]

pe0mgb, 2023-03-19:
Fabio,
An other common failure is to have more installations of TFT_eSPI library. Check that. And take a look at the link-list of the IDE compiling process. There you can see which library is used in the linking proces.

Verzonden vanuit Mail voor Windows

Van: FabioLucca01
Verzonden: zondag 19 maart 2023 19:06
Aan: pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino
CC: Gert PE0MGB; Comment
Onderwerp: Re: [pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino] TFT rotate (Issue#23)

Hi, my "User_Setup_Select" is correct like yours above:
#include <Setup1_ILI9341.h> // Setup file configured for my ILI9341 2.8 inch esp32
The ".ino" sketck is correct, I used a standard example from the TFT_eSPI library and the display doesn't work right, always cutting off some part of the screen, I will study the library configuration files and make some changes to see if it fixes it, if I find it the solution I return here. Thanks for the help.

FabioLucca01, 2023-03-21:
Hello, I managed to make it work just by changing 3 lines of code in the .ino . One in the constructor forcing it to the dimensions of the screen (320x240) and the others were referring to the orientation (rotate(2)) and the touch that was changed. Attached image with changed lines.
It's certainly not the most elegant way to solve it, but that's how I managed it.

Grateful.

[image]
[image]

pe0mgb, 2023-03-21:
Good to hear that it is working. Still confused about your solution and why. There are already hundreds of radios build whitout any problem. But anyway it works. Did you changed the color of the buttons etc.All colors are different.

FabioLucca01, 2023-03-21:
Hi, I only changed 3 lines of code from the .ino that are in the above comment image.
My libraries are up to date and I recently downloaded all of them. If the colors are different, it wasn't me who changed it, so I suspect that this display is a clone or not an ILI9341 and therefore has some differences... but it ended up working. Now I'm going to work on the RF part, I'm waiting for the Si5351 to arrive.

Once again thanks for the support.

pe0mgb, 2023-03-21:
Clear, thanks.