pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #24: Compilation error: 'class TFT_eSPI' has no member named 'setTouch'

andreasmon, 2023-06-13:
Hello,
 i try to verify code form here https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino
doit devkit v1 with tft 240x320 2.8" SPI TFT LCD Touch Panel Serial Port Module With PBC ILI9341.

i run version SI473X_2.8_TFT_V3.3          

I get errors but fix enough.
Now i get error :

error: 'class TFT_eSPI' has no member named 'setTouch'

i try many ,but nothing fix it.

Any help;
Thank you.

andreasmon, 2023-06-13:
First time try verify the code, i fix some errors by add  thef ollow lines in code:

#include <Adafruit_GFX.h> 
 #define TFT_CS 15

pe0mgb, 2023-06-14:
First of all there are no errors in the code. Many V3.3 programs work to complete satisfaction. V3.5 has more significant improvements in operation and presentation. First remove all added include and defines. They only cause more misery.

Start by reading the readme file on the opening page of the SI4735 in my GitHub. Copy the file Setup1_ILI9341.h to the root directory of TFT_eSPI library and replace the User_Setup_Select.h file in the same root directory. Make sure you have the latest version of this library installed.

Now the mentioned compiler errors will be a thing of the past.

Greeting,

Gert

andreasmon, 2023-06-14:
Hello.thank you for your time and your answer.

I have 7 days trying compile the code.I have read the readme file and do things you write about.
I also have install last version  2.0.9  of esp32 by espressif systems.

i delete libraries and files and start again to compile the code of version 3.3,but i get error again.
I try it to another computer that i install arduino ide for first time and same error again.

Last i  record a video for this , installed the libraries and try to compile the code .I upload it to youtube .Its free to watch.

Can you, when you have time watch the video and  find what i am doing wrong; Or maybe my english not help me understand correct what i must do .

link for free watch youtube video of error:

https://www.youtube.com/watch?v=I0TgrD7qIwc

Thank you again.

pe0mgb, 2023-06-14:
Maybe you have installed the TFT_eSPI lib. twice.

https://www.youtube.com/watch?v=I0TgrD7qIwc

pe0mgb, 2023-06-14:
I'm traveling at the moment . I will react in the coming weekend.

https://www.youtube.com/watch?v=I0TgrD7qIwc

andreasmon, 2023-07-02:
hello,

did we have any news;

thank you.

pe0mgb, 2023-07-05:
Hello,

Sorry for long delay but I'm traveling and not always in contact with a
network.

I have tried to follow your youtube info.

Not always easy but anyway. I think that I have seen that in the top
directory of TFT_eSPI library the wrong files are installed, checking the
date.

Please install my two files from github in the top directory from TFT_eSPI.

The no touch compiler error is caused because the touch info is missing in
the default setup1_ ILI9341.h file. My files solve that problem.

Hope this will work.

Success

 

<https://github.com/pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino/issues/24#
GD2DANCNFSM6AAAAAAZFMPBEQ> .
om>

andreasmon, 2023-07-06:
thank you for your help and also understand that you are working...So no problem for the delay.

I do not understand who files of the library,i must replace .If you check it and working can you send me the working .ino file aarduino code  or also and the correct library;

Thank you.

andreasmon, 2023-09-15:
hello,i finnaly solve the problem and upload code.I upload version 3.3

Now touch screen ILI4931 show the above message:

SI4735/32 RADIO
VERSION 3.3
02-11-2021
SI4735 not detected

and program stop,freeze there...

in my circuit i have used sis4732 ic,and no the  4735.

i upload also version 3.5,but same again