pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #25: SI5351 not found

dpechman, 2023-12-23:
I have tryed SI4735_2.8_TFT_SI5351_V3.5 with crystal and it runs nice. 

Have then tried to use si5351 enabling it on:

//#define IhaveCrystal
#define IhaveSI5351

but its returning SI5351 not found and its not generating any frequency on output. 

How to debug it?

dpechman, 2023-12-23:
I did a test inverting data and clock (on pink board) and it starts without msg of **SI5351 not found** and generate a frequency about 42MHZ on output, but frequency generated not change at all.

Using the right pins the output is dead.

73

pe0mgb, 2023-12-24:
Daniel,

Hereby a picture how the si5351 is connected to the radio. When it is not running it is hard to find out by me what’s going wrong. The software is OK and in use by many many people.

Do you have placed the jumper, left of the coil at the picture, in the right position?

When you still have problems please let me know.

Best regards,

Gert PE0MGB