pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #27: ledcSetup error ?

z34pec, 2024-09-08:
Hello. 
I am trying to compile latest 3.5 version (my ESP 32 is found as Lilygo T-display).
When compiling I receive this error: Compilation error: 'ledcSetup' was not declared in this scope

Above this error I see where it exist: 

'ledcSetup' was not declared in this scope
 968 |   ledcSetup(LedChannelforTFT, LedFreq, LedResol);
'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'? 
969 |   ledcAttachPin(Display_Led, LedChannelforTFT);

How to solve this problem ?

73 de Z34PEC

z34pec, 2024-09-08:
I also tried all of given versions. Also copied suggested files to required TFT_Espi library (one in root of the library and the second in Users_Setup folder). Still receive thess ledcSetup erro while trying to verify your sketch. Where I go wrong I do not know, I lost  a day trying to figure out what is the error.
Here is screenshot of error.
[image]

z34pec, 2024-09-08:
OK, looks like I find the cause of this error.
At thismoment latest update version of ESP 32 boards from espressif is 3.0.4 and error exist in every version that start with 3.x
With 2.0.17 there is no error but no communication is make with esp32.
Looks like there should be update of sketch to work with newest esp32 boards updates.

garrysblog, 2025-04-23:
In the ESP32 v3 core ledcSetup and ledcAttachPin have been merged into the single command ledcAttach. I changed these lines in SI4735_2.8_TFT_SI5351_V3.5.ino from this:

```
ledcSetup(LedChannelforTFT, LedFreq, LedResol);
ledcAttachPin(Display_Led, LedChannelforTFT);
```

To this:

`ledcAttach(LedChannelforTFT, LedFreq, LedResol);`

The sketch compiles and uploaded without errors and appears to be running correctly. However, I'm not particularly familiar with all the features of this radio so am not 100% sure there are no other side effects of using the v3 core.

There is more about the ledcAttach change here [https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ledc](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#ledc)

I am using Arduino IDE v2.3.6 and v3.2.0 of the esp32 board core library