pe0mgb/SI4735-Radio-ESP32-Touchscreen-Arduino #29: Error Compilation

cabralcr, 2024-12-03:
[image]

Hello, I'm trying to compile version 3.5 and this error is appearing, and all the files are in the same directory and I've already replaced the file in the main library, could anyone help me?

ta3lg, 2025-01-05:
[image]
Merabalar, derleme atası bende de var.Gidermeye alışıyoruz. Bu sorunları giderebilen varsa yardımcı olmalarını bekliyoruz. Teşekkür  ederim.TA3LG/Ramazan Turkey

RecepAGulec, 2025-01-18:
Hâlâ derleme problemi devam ediyorsa, Arduino IDE'de ESP32 kartının sürümünü kontrol et. Eski 2.0.17 sürümü ile değiştir.